TweetyDaBird/Lotus-58-Classic #6: Latest version

fi0, 2021-03-09:
Which version should I build?

azzamsa, 2021-03-10:
Yes, According to the timestamp `5 days ago`. `v1.06` is the latest release.

TweetyDaBird, 2021-03-12:
The plates and PCB's are in separate packages, since you cannot order them as one item from a manufacturer (one order, but multiple items). As such, they have separate version numbers.

So for a complete keyboard you'd need the v1.1 PCB's and bottom plates plus whichever top plates you pick to go along with them, from either package of plates.

fi0, 2021-03-12:
Thank you! I was confused with PCB versions and plate versions. I should have looked into https://github.com/TweetyDaBird/Lotus58/releases instead of the tags.

closed 2021-03-12 by fi0