TweetyDaBird/Lotus-58-Classic #10: LED pinouts not the same for SK6812MINI and SK6812MINI-E

mwmccarthy, 2021-05-16:
The pinout for the SK6812-MINI matches the orientation of the PCB for soldering from the bottom with the lens facing away from you as you solder:

[image]

However, the SK6812MINI-E would need to be soldered from the top of the PCB with the lens facing up in order for the VDD and GND pins to line up, but then DIN and DOUT are swapped:

[image]

TweetyDaBird, 2021-05-17:
Well, that's altogether stupid that I hadn't verified that part. I have only built with wigless, and figured (without checking) that tehy had the same pinout.

Removing them from parts list.

closed 2021-05-17 by mwmccarthy

phil123456, 2024-05-06:
there are so many pinouts of the mini version, which one is correct ? does this depend of the manufacturer or are they all the same ?

TweetyDaBird, 2024-05-06:
> there are so many pinouts of the mini version, which one is correct ? does this depend of the manufacturer or are they all the same ?

Each standardized model has a pin out. That's the same from whatever seller, as they all come form the same producer. But no, mini and mini-e are not the same, and never will be.