TweetyDaBird/Lotus-58-Classic #17: Where is the breakaway part for when you order from JLCPCB?

SuperNici, 2021-08-12:
You mentioned [here](https://github.com/TweetyDaBird/Lotus58#things-to-note-when-ordering-pcbs) that you have added a breakaway part on the PCB Gerber files. 

> If ordering from JLPCB the plate Gerber files include a reference putting the added serial etc. on a breakaway part for a clean look with minimal cost, if using another manufacturer it's possible they have additional fee's for removing the extra text, or you should consider plotting the Gerber files yourself with the correct reference for your manufacturer.

I've been looking for it for a while and couldn't find it.

Hybrid-Force, 2021-08-16:
Hey there, I think the breakaway parts were removed in commit 5263f9b0050a36a8181bea90f3474fcbf5af5144. The reference text for serial number was originally on the breakaway part of the upper thumb key.

TweetyDaBird, 2021-08-23:
This depends on if you pick the plates with breakaway parts in the switch holes or not. The plates are available both ways to allow it to work with all PCB manufacturers. 

Usually the text is small an unobtrusive enough that it’s hidden under the keycaps, but if you choose to order the PCB’s yourself, it’s worth to check and re-render as needed. Or pay the usually minimal fee to have the text removed. My comment was more aimed at making you aware of this, than an exact roadmap to how to order. 

BTW, if you prefer to not bother with ordering yourself, I’m selling PCB kits on tindie at what I think are pretty reasonable prices.

SuperNici, 2021-08-23:
Appreciate the comments @Hybrid-Force and @TweetyDaBird.

I feel like playing the small extra fee on jlcpcb will be worth it.

>BTW, if you prefer to not bother with ordering yourself, I’m selling PCB kits on tindie at what I think are pretty reasonable prices.

You should absolutely put that in the read me! I would have never known! Though for myself i will (soon) order the parts myself because I'll be adding some small stuff to the pcb.

And to support these kinds of open source projects ill also make sure to donate a small amount to you three main contributors :)

closed 2021-08-23 by SuperNici