TweetyDaBird/Lotus-58-Classic #20: RGB leds power draw problem

mattscooter, 2021-10-05:
Firstly thank you for your work on this.  You've made everything so easy to order with links to the Aliexpress.  I really appreciate it!

I recently finished soldering everything and have run into an issue with the RGB leds.  Note that I have wired up the 58 individual keys but not the underglow leds. 

To get the firmware built I took the lotus58 folder from the keyboard folder in your qmk_firmware fork and put it into my local qmk environment (running v14.19).

With the RGBs disabled the keybord is working fine.  With the default in your config of 6 leds on each hand everything is working fine.

The issues started when I enabled all RGB leds on both sides. I managed to get everything lit up and could briefly change colours but then after a short while the half that's plugged in would keep working for a while and the other half would stop.  Then after a little while longer the keyboard stopped responding altogether.

If I enabled only 34 leds - everything is stable.  When I enable more than this the system started acting up as above.  It doesn't matter how many leds are enabled on each hand I could have all of one hand lit and only a few on the other such as RGBLED_SPLIT { 29, 5 } or an even number on each hand.

After much testing I have identified that I can enable all of the leds but it's the brightness that's causing the issue.  If I restrict the brightness to a maximum of 125 I can get all LEDs lit on both hands.  Does this mean that the system is failing because it's trying to draw too much power to light all of the leds at full brightness?

So I have a work around in setting the max brightness to get everything working but I was wondering if it's possible this is an issue with my RGB led circuit that could be causing the problem?  Have you run into this problem before? If not do you have any tips for how I might be able to debug it?

TweetyDaBird, 2021-10-05:
Hi Matt,

That’s not even a bug/problem, that’s just simple math!

The RGB LEDs use between 4 mAh and 50 mAh each depending on brightness. 58 of them at full brightness is 2900 mAh and the USB port can only supply 500 mAh. 

The design was never meant to support all 58 RGB’s at full brightness, as that would be outright blinding. Also, the RGB’s was chosen for the ability to have great colors and light at less than full brightness and work with the power limits. 

Using the under-glow though, you can ramp up the brightness fairly high, and it’s more useful that way, as the light is indirect. That’s why I didn’t include the brightness limit as it was configured. 

What happens is fairly simple really. As the brightness ramp up on the RGB’s on the USB supplied hand, the voltage drops over the TRRS and the slave side stops working as it plain and simple crashes. 

Setting the brightness limit to ~100 isn’t really a workaround, but the correct and intended way to do it. 125 ish would work as long as you don’t set the color to white, or use the type of effects that changes the color too fast. 

So sorry, but your keyboard is working perfectly, no debugging needed!

/Markus

mattscooter, 2021-10-07:
Thanks for the explanation!  I suspected that this was what was happening after I did all of my testing. 

I'm glad to hear that my keyboard is working as expected.  Now I just have to design and print a 3d case for it and I can finish the build off!

By the way I'm starting to try and configure the RGB led matrix.  This means defining the g_led_config with the appropriate led indexes and positions.  Have you, or do you know if anyone else has already done this?

TweetyDaBird, 2021-10-07:
There are plenty of people that have configured it, but AFAIK none of them have shared the code (as of yet). 
And I admit to being lazy, once I had verified that it all worked, I decided to stick with under-glow for my at-glance layer indications, and didn't really bother with the rest.

mattscooter, 2021-10-09:
Fair enough. 

I've had a bit of an attempt and I'm pretty happy with the results so I'm happy to share my code here in case someone else is looking. 

```
 led_config_t g_led_config = { 
  {
        // left hand switch matrix to led map
        {  5,  4,  3,  2,   1,   0 },
        {  6,  7,  8,  9,  10,  11 },
        {  17, 16, 15, 14,  13, 12 },
        {  18, 19, 20, 21, 22,  23 },
        {  NO_LED, 28, 27, 26, 25, 24 },
        
        // right hand switch matrix to led map
        {  40, 39, 38, 37,  36,  35 },
        {  41, 42, 43, 44,  45,  46 },
        {  52, 51, 50, 49,  48,  47 },
        {  53, 54, 55, 56,  57,  58 },
        {  NO_LED, 63, 62, 61, 60, 59 }
    },
    {
        // Left Hand switch led positions
        {86,0},   {69,0},   {52,0},   {34,0},   {17,0},   {0,0}, 
        {0,16},   {17,16},  {34,16},  {52,16},  {69,16},  {86,16}, 
        {86,32},  {69,32},  {52,32},  {34,32},  {17,32},  {0,32}, 
        {0,48},   {17,48},  {34,48},  {52,48},  {69,48},  {86,48}, 
        {103,40}, {103,64}, {78,64},  {60,64},  {43,64},

        // Left Underglow led positions
        {52,64},  {86,48},  {9,48}, 
        {9,0},    {43,0},   {78,0}, 

        // Right Hand switch led positions
        {138,0},  {155,0},  {172,0},  {190,0},  {207,0},  {224,0}, 
        {224,0},  {207,16}, {190,16}, {172,16}, {155,16}, {138,16}, 
        {138,32}, {155,32}, {172,32}, {190,32}, {207,32}, {224,32}, 
        {224,48}, {207,48}, {190,48}, {172,48}, {155,48}, {138,48}, 
        {121,40}, {121,64}, {146,64}, {164,64}, {181,64},

        // Right Underglow led positions
        {172,64}, {129,48}, {215,48}, 
        {215,0},  {181,0},  {146,0}
    },
    // led flags
    {
        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT,
        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT,
        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT,
        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT,
        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, 
        
        LED_FLAG_UNDERGLOW, LED_FLAG_UNDERGLOW, LED_FLAG_UNDERGLOW, 
        LED_FLAG_UNDERGLOW, LED_FLAG_UNDERGLOW, LED_FLAG_UNDERGLOW,

        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT,
        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT,
        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT,
        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT,
        LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT, LED_FLAG_KEYLIGHT,

        LED_FLAG_UNDERGLOW, LED_FLAG_UNDERGLOW, LED_FLAG_UNDERGLOW, 
        LED_FLAG_UNDERGLOW, LED_FLAG_UNDERGLOW, LED_FLAG_UNDERGLOW
    }
 };

```

But now I think I've gone way off topic and you've answered my initial question so I'm going to close out this issue!

closed 2021-10-09 by mattscooter