TweetyDaBird/Lotus-58-Classic #25: jumper pad questions

nathish, 2021-12-05:
Just received the v1.21 pcbs and have a couple of questions regarding the jumper pads. I'm assuming I need to jump one of the pads to the middle to get the LEDs to work but I'm not sure which. 

I have the per key and underglow LEDs soldered already, with the intention of using both. Can you let me know what pads I should jump to get all LEDs to work on both hands?

[image]
Also, I'm curious as to what these pads are used for. Appreciate that this is a pre-release so I'm asking for the docs that have yet to be published. thanks in advance.
[image]

TweetyDaBird, 2021-12-06:
Let’s start backwards, it’s that kind of day for me!

1. Reset, easily accessible on both sides for a screwdriver or whatever. 

2. Designates left/right hand. 

3. If you close ‘per key’ RGB AND this jumper, you create a full string of the RGB for it all to work. Not really recommended for high output/high power draw, but feasible at low intensity.

nathish, 2021-12-06:
Perfect, thanks for the quick response. one more question before I solder the battery switch. below switch S1, the plus and negative pin holes, just want to double check that these route to the power and ground of the MCU (nicenano v2 in my case)?

TweetyDaBird, 2021-12-07:
They route to the switch and then the raw/batt+ pin and GND/batt- so meant for the battery.

closed 2021-12-07 by nathish