TweetyDaBird/Lotus-58-Classic #26: Breakaway tabs referenced in readme

mrherman, 2021-12-10:
The ReadMe has a section titles "Things to note when ordering PCB's" but looking at the kicad_pcb files in the Plate directory I don't see any with breakaway tabs.  I do see a jpg in the same directory that shows a pictiue of a rendering of a board with breakaway pads.  Do pcb files of these plates exist (am I just looking in the wrong place)?  Do I need to do something to generate them?

TweetyDaBird, 2021-12-11:
Yes, you are likely just looking in the wrong place. 

You can generate the files from KiCad yourself if you like, the data is there if you look for both with and without tabs. 

But if you’re not looking to modify things, just go to releases, and you’ll find the gerber files.

closed 2021-12-11 by TweetyDaBird

mrherman, 2021-12-16:
I didn't see the with tabs in KiCad.  I looked at both branches.

TweetyDaBird, 2021-12-17:
Well, since the plates without tabs were generated last and from the ones with tabs, the saved code in the open development is just as you say without the tabs. 

If you however look at the snapshot of code from when the plates with tabs were generated (included as a zipped file in the release) that's the exact data it was generated from, and thus without tabs. 

Its also possible to find the exact commit where the same data exist by looking in that history, but since you have a release to go by, that would be the easiest way. 

However, I'm genuinely curious what you want the PCB file for that you can't do with the gerber? Manufacturing the plate as a PCB will need the gerber, and editing the PCB will generate a lot of errors in the DRC as it's not a complete PCB with a complete schematic, so less trivial than the actual PCB infact.

mrherman, 2021-12-17:
Thanks Markus.  It didn't dawn on me that a commit would over-write the with tabs with without tabs.  I was expecting two separate files.  I can see them now.

I wanted the PCB files because I am making a small change to the layout.  I am familiar with KiCad that I shouldn't run in to problems.

Thanks for the help.

TweetyDaBird, 2021-12-17:
Well, if it was a full on PCB I would obviously have made it as separate files etc. but for just retaining the outline and such to make a plate or versions of it, I was a little lazy.