TweetyDaBird/Lotus-58-Classic #27: Issue with communications

NarenHT, 2021-12-27:
Hi,

I am having issues with the communication between the boards.
I have made the older Lotus58 v1.11. I have only modified your firmware in the keymap.c file and the lotus58.h file to use the default lily58 keymap instead. Everything else is identical to your firmware. I have closed serial and added i2c resistors. I have an oled on the right and an encoder on the left. Keys everywhere else. TRRS cable shows continuity.

My primary issue is that only the board connected to the USB works. I am sometimes able to start the display on the right, when the usb is connected to the left but it will only display the boot image at best. 

Both my power jumpers are open. Is this the problem? Should I always connect to the right and close the right power jumper?
Something like this: https://docs.splitkb.com/hc/en-us/articles/360010588860-Only-one-half-of-my-keyboard-works-at-a-time-but-not-when-they-are-both-connected

Your help is appreciated!

NarenHT, 2021-12-27:
Also, I have no perkey / underglow leds.

TweetyDaBird, 2021-12-27:
Ok, let’s start to unravel this, as some things you are saying aren’t making complete sense for me combined together.

The i2c resistors are only needed on the side using the OLED. And if you are closing the serial jumpers, check that you are in-fact using serial in the .h file. Both options should be in there as it’s optional, can’t remember which is commented out as default, but I think serial will be the active. Also make sure you haven’t closed the i2c jumpers as well, then it’s surely out of whack. 

The power jumpers (if we are talking about on the ProMicro) should normally not be closed, but random clones may have other uses for that jumper than default, so double and triple check what it does. 

One thing I notice is you didn’t mention if the two sides work/identify as their respective sides when used with USB (ie the correct keys, not mirrored). The way my firmware is set up, if you solder the left/right jumpers, they should work as their respective sides, and then you need to double check that you are connecting USB to the side designated as master. (Connecting to the slave works, but not well). To save on cross traffic, I’d set the OLED side as master.

NarenHT, 2021-12-27:
Hey! Thanks for the fast reply.

i2c are only on the OLED Side. I am using Serial by default. I have commented out the #define USE_I2C for the moment.
The right side has OLED and is set to default master. However, detect USB is enabled also.
Two sides work perfectly individually.
Let me check the solder connections once in the morning. I think I have messed something up then.

p.s. Here's a pic for you! Thanks for the project.
[image]

TweetyDaBird, 2021-12-28:
Thanks for the pic. 

Well, so there is nothing unclear. To use the OLED you do not need to solder the jumpers by TRRS. The i2c lines are already connected.

NarenHT, 2021-12-28:
I might have misunderstood something. Can I close both serial as well as i2c jumpers? Because that's what I have done.

TweetyDaBird, 2021-12-28:
No, most definitely not. That’s causing a short circuit and would be the reason for your issues. 

Luckily though, all involved pins should survive the experiment if nothing exceptional has happened.

closed 2022-01-10 by TweetyDaBird

NarenHT, 2022-01-10:
Thanks. Keyboard is fixed. I forgot to update this. Only noticed that the display will not show data after boot image if the non display side is connected to PC. Otherwise all cool. Keys work and that's all that matters to me. Thank you.

TweetyDaBird, 2022-01-10:
That's most likely because the display is only active on the master (should be the one normally connected to USB/with display) in the firmware. You can tweak it to work regardless, or have different functions depending on which side is plugged in etc. but that's beyond what the basic firmware is intended for, and up to you to fiddle with.

NarenHT, 2022-01-10:
Absolutely. Cheers!