TweetyDaBird/Lotus-58-Classic #28: Can't find what to short to use serial (v1.21)

wigeric, 2022-01-04:
Hi, 
thanks for all your amazing work but I'm unable to find what jumper to short for using serial (2 oled screens).

The wiki says
`If using Serial, close the one single jumper above the TRRS socket on both PCB's Required when using two OLED screens`

But the only things I'm able to see above the TRRS socket are : 

- D30 for encoder
- Jumper for hand
- Jumpers for OLED

Could you point me in the right direction ?

TweetyDaBird, 2022-01-04:
Actually, I can’t. That wiki/build guide is for the v1.x versions only. And the reason it’s not yet updated is that you are building from a prerelease. 

And while the particular release you used was removed because it has other issues, in its description there was a hint. The PCB no longer has any jumpers for i2c/serial since it no longer supports using i2c for the TRRS. It’s hardwired to serial and the displays are wired up to i2c. No conflicts that way.

wigeric, 2022-01-04:
Thank you, that answers my question.

closed 2022-01-04 by wigeric