TweetyDaBird/Lotus-58-Classic #29: How to test the keyboard ?

Araly, 2022-01-06:
Hey,

I've been building this keyboard as my first soldering project, and I'm pretty sure I've done several mistakes in it. Two of the breakaway bridges broke (the left-hand indicator on the left board, and the lower i2c bridge on the right board) and I soldered them back, but I'm not sure if this would actually work. I'm trying to understand if I've ruined the boards or if it's salvageable, and how I could test the connections. I'm also trying to understand how to compile / configure / flash the firmware once I'm done soldering the microcontrollers.

I've socketed the first controller, but I find it very hard to remove it from its socket. I'm able to leverage on its back, but I can't find a way to pull it off its front where the USB port is, as I don't want to damage it.

Is there a way I can see what is supposed to connect with what ? I'm thinking of buying a multimeter to see where if the bridges that are broken are still making a connection or not. A friend said to wire things together to bypass the bridge if they are ruined.

I've found this repo https://github.com/TweetyDaBird/qmk_firmware that you seem to mention in another issue being the forked firmware to use for the board, but looking around I couldn't find how you're supposed to compile it or what you need to do to flash it. I also see that the section for it is missing in the wiki. Is there maybe another set of instructions I can follow and adapt to Lotus ?

The issues I had with the bridges were after soldering them together with neighboring things, and trying to disconnect them by cutting the excess of solder between them. It seems like the tip I'm using on my soldering station is too thin and doesn't transmit heat well, making me have to use the side of the tip instead, and being more messy with the soldering. So I'm thinking of buying larger or differently shaped tips instead.

Thanks

Araly, 2022-01-07:
I see, I understood that Serial was to be used for when you had a special case, can't rememeber if it was two OLED screens maybe, so I assumed i2c was the default recommended one. I guess I could remove the solder from the i2c bridges and go Serial instead.

I'll try the scrapping off method to repair the LH left hand bridge indicator.

Thanks for the quick reply ^^

TweetyDaBird, 2022-01-07:
A few versions back, i2c was the preferred option, with lower latency etc. A couple of rewrites of the code later, it really doesn’t matter much, and serial gives more options. Essentially you can pick either or.

Araly, 2022-01-07:
using a multimeter I was able to check what worked and what didn't, scraped off one of the bridges and replaced it with some diode leg instead. Multimeter beeps where I expect it should so that seems fine.

I'm trying to compile the firmware now, I've installed qmk with `yay -S qmk` and I cloned your fork. I copied the `keyboard/tweetydabird` to the qmk's own keyboard folder, and tried to run `qmk compile -kb tweetydabird/lotus58 -km default`, but that failed

```
$ qmk compile -kb tweetydabird/lotus58 -km default
Ψ Compiling keymap with make --jobs=1 tweetydabird/lotus58:default

QMK Firmware 0.15.12
☒ Not including data from file: keyboards/tweetydabird/lotus58/info.json
☒       layouts.LAYOUT: Additional properties are not allowed ('key_count' was unexpected)
Making tweetydabird/lotus58 with keymap default

☒ Not including data from file: keyboards/tweetydabird/lotus58/info.json
☒       layouts.LAYOUT: Additional properties are not allowed ('key_count' was unexpected)
avr-gcc (GCC) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/tweetydabird/lotus58/keymaps/default/keymap.c                                 In file included from quantum/action.h:23,
                 from quantum/keymap.h:22,
                 from quantum/quantum.h:21,
                 from keyboards/tweetydabird/lotus58/lotus58.h:19,
                 from keyboards/tweetydabird/lotus58/keymaps/default/keymap.c:16:
quantum/keycode.h:100:17: error: redeclaration of enumerator 'KC_NUM_LOCK'
  100 | #define KC_NUM  KC_NUM_LOCK
      |                 ^~~~~~~~~~~
keyboards/tweetydabird/lotus58/keymaps/default/keymap.c:27:5: note: in expansion of macro 'KC_NUM'
   27 |     KC_NUM,
      |     ^~~~~~
quantum/keycode.h:328:5: note: previous definition of 'KC_NUM_LOCK' with type 'enum hid_keyboard_keypad_usage'
  328 |     KC_NUM_LOCK,
      |     ^~~~~~~~~~~
keyboards/tweetydabird/lotus58/keymaps/default/keymap.c:204:6: error: conflicting types for 'oled_task_user'; have 'void(void)'
  204 | void oled_task_user(void) {
      |      ^~~~~~~~~~~~~~
In file included from quantum/quantum.h:180,
                 from keyboards/tweetydabird/lotus58/lotus58.h:19,
                 from keyboards/tweetydabird/lotus58/keymaps/default/keymap.c:16:
drivers/oled/oled_driver.h:290:6: note: previous declaration of 'oled_task_user' with type '_Bool(void)'
  290 | bool oled_task_user(void);
      |      ^~~~~~~~~~~~~~
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:457: .build/obj_tweetydabird_lotus58_default/keyboards/tweetydabird/lotus58/keymaps/default/keymap.o] Error 1
Make finished with errors
make: *** [Makefile:478: tweetydabird/lotus58:default] Error 1
```

compiling the example keyboard seems to work though. I'm following this guide: https://docs.qmk.fm/#/newbs_getting_started

Araly, 2022-01-07:
was able to get it working by using the entire fork instead of just copying the keyboard/tweetydabird/lotus58 directory. I also needed to install qmk with python instead of pacman, as the avr-gcc was too new, and it made the firmware too big

TweetyDaBird, 2022-01-07:
Well, it seems I need to update it to the newest version then.

Araly, 2022-01-07:
I'm currently running into an issue where 
```
Device /dev/ttyACM0 has appeared; assuming it is the controller.
Waiting for /dev/ttyACM0 to become writable..........................................................................
```

I see that I don't have the rights to write to the device, is there a thing I'm missing ? It doesn't seem like the device belongs to any groups I can add my user to, and I can't run `qmk flash` with sudo

Araly, 2022-01-07:
needed to run the recommended command in `qmk setup` reguarding udev, was able to flash one side

TweetyDaBird, 2022-01-07:
Unless you have wondered both sets of jumpers, I’d assume your problem is firmware, since it is set up for serial, so yes you’d need to change most of that. 

Correct on switching master.

Araly, 2022-01-08:
I soldered both I2C jumpers on both sides, and I haven't soldered the serial jumpers. Where do the I2C jumpers connect ? I'd like to be able to test that they do connect properly.

What should I change ? I tried commenting out the serial line and replacing it with USE_I2C, but when I flash that nothing works anymore, not even the sides individually. Do I need to also remove the other serial lines, or replace them with I2C related lines ?
```c
// #define USE_SERIAL
#define USE_I2C
```

TweetyDaBird, 2022-01-08:
They rather obviously connect between the TRRS and the ProMicro. 

Have a look at QMKs docs. It’s outlined step by step making the settings for i2c or serial. 

The resistors are not exclusively for the OLED, but needed for i2c period. 

Good policy to things step by step.

Araly, 2022-01-08:
if the resistors are needed for i2c, and are not exlusively for the OLED, should I have them on both boards ? I only have them on the left board, where I want to put the OLED

TweetyDaBird, 2022-01-08:
Even if you already fixed it, for completeness, no they are only needed once per i2c system.

Araly, 2022-01-09:
just like mentioned in this diagram yeah https://docs.qmk.fm/#/feature_split_keyboar

I have one more question. I bought an encoder that I want to place on the high corner, above where the controller is. It has metal legs that are said to be snipped off, but it also has bumps like the switches on the bottom. Unlike the switches though, there's no hole for those bumps to go in. Should I also snip off these bumps ?

[image]
d
[image]

Araly, 2022-01-09:
I've cut them off, everything seems good so far

TweetyDaBird, 2022-01-12:
Those are new, I haven't seen them on any encoders before.

Araly, 2022-01-14:
[image]

keyboard looks good and works well, now onto making a comfortable keymap ^^

closed 2022-01-14 by Araly