TweetyDaBird/Lotus-58-Classic #31: corrupt plate dxf files?

sungo, 2022-02-03:
I'm looking to laser cut some plates (new glowforge gotta forge) but it looks like the plate DXF files are corrupt. Peaking inside, it appears the DXFs have gotten wrapped in github html/js fun somehow. Any chance of getting clean DXFs?

TweetyDaBird, 2022-02-03:
Sorry, but those files were never made or maintained by me but by another user. I have no way of recreating them.

sungo, 2022-02-03:
Alrighty. I'll see what I can come up with. If I figure something out, do you want a PR?

TweetyDaBird, 2022-02-03:
Gladly!
While personally I'm quite happy with the FR4 plates, it seems a lot others want to use different materials, hence why they were uploaded here. So I'm happy to take whatever files you come up with.

sungo, 2022-02-07:
Looking through git history, it looks like the corrupt versions were added here. https://github.com/TweetyDaBird/Lotus58/commit/1c34fa41cd076e58e1640e762ec52977901df85a. They appear to be a "save source" of github's HTML view of the original DXF files. I extracted the original files from git history and they import into my tools fine. Haven't tried cutting them yet. 

If your intention was to bring those dxf files forward with no changes, I can drop a PR with the original files. Which branch would you like them to land in?

TweetyDaBird, 2022-02-09:
Probably 1.11 branch. Yep, the intent is to keep files intact.

closed 2022-02-19 by TweetyDaBird