TweetyDaBird/Lotus-58-Classic #38: Specify Microcontroller orientation

SuperNici, 2022-07-02:
[Here in the wiki](https://github.com/TweetyDaBird/Lotus58/wiki/3.-Building-the-basic-keyboard-circuit#step-5-controller-sockets-and-optional-reset-switch-or-optional-battery-cut-off-switch) it is unclear whether the controller should be mounted facing up, or down. 

Pictures would also help a lot.

TweetyDaBird, 2022-07-02:
Easily addressed.

closed 2022-07-02 by TweetyDaBird

SuperNici, 2022-07-02:
> Place the controller facing the main PCB, on the the backside/component side of each PCB, if socketed flush with the socket, if not socketed place it level as the USB port touches the main PCB.

My bad, its been a long day of troubleshooting ( ; U ;)b

TweetyDaBird, 2022-07-02:
I just edited to add that, so no error on your part.

SuperNici, 2022-07-03:
Ah, Id also like to add that the orientation is only written on the v1.23 version of the PCB:
[image]

I have the 1.11 one here which doesnt mention it:
[image]