TweetyDaBird/Lotus-58-Classic #52: Short between vcc and gcc

Koshu, 2023-09-12:
Hi,

I am currently building your board and found a short in my PCBs between GND and VCC and in the release gerber files version 1.33. 

Looks to me as if the GND plane has not been updated after moving a trace. 

Gerber viewer:
[image]

In PCB:
[image]

Guess I will now figure out how precise my pcb scratching abilities are 😅. Otherwise nice board btw, first buttons are already working.

TweetyDaBird, 2023-09-12:
Crap. My apologies. 

The file will be regenerated/upgraded asap.

Koshu, 2023-09-12:
Don't worry about it, I was able to fix it. Just the usual stuff in DIY projects ;). 

Did a few changes because I decided to use a sparkfun rp2040 for some reason but the left hand seems to be working now. 

[image]

TweetyDaBird, 2023-09-13:
Well, it still shouldn't happen. File is now updated.

closed 2023-09-13 by TweetyDaBird