TweetyDaBird/Lotus-58-Classic #62: Version 1.30, 1.31 contain incorrect readme files?

Catacomba, 2023-12-23:
I ordered the 1.33 (at the time latest) version PCB, and when searching for the parts list, because the 1.33 version of the readme does not contain the parts list in the readme, i backtracked the versions. Most recent version to contain the parts list is 1.31. Which i followed when buying parts.

However the readme in folder 1.31 is actually has v1.23 specified in the title (I overlooked this... and used the readme just for the parts list). The same readme also appears to be in 1.30. 

These readmes are actually for the TRRS version of the Lotus... and i ordered the USB-C PCBs... I know this is my mistake, because I did not pay attention to the release notes and didn't realize this changed with v1.3.

However I believe that the readmes in 1.30 and 1.31 should be fixed with the correct parts lists or perhaps the TRRS row removed from the parts list or maybe just a simple strike trough, so that it doesn't mislead other people.

TweetyDaBird, 2023-12-23:
Well, unfortunately you overlooked even more than that. Neither v1.30/1.31 or 1.33 reached a full release until 1.33 was moved to that status about three-four weeks ago.  Until then all three were pre-releases (and some still are). And that label signifies just that. Not polished, not fully documented, not ready for public consumption _unless_ the user takes care and either asks questions where applicable, or knows to double check and question info. 

And never ever will I take any responsibility for older documentation not matching another version. That's just not the correct way to interpret it. 

Advice for the future. Asking questions is a lot better than assuming the answer. 

In fact unless you ordered the PCBs in that timeframe, the PCBs may not even have been build tested? (A pre release becomes a full release once build tested). I'm not sure on the timeline here. 

And since the releases have been improved (with all but the latest) with tiny little changes that doesn't change the function much, but makes it easier to build or fit a case etc. they will never be full releases, as that will be v1.40 which is intended to be stable for a while. 

And as such, it's in the process of getting updated and completed documentation. That always will happen _after_ the build test, and after it reaches a full release. The tags actually matter.

Catacomba, 2023-12-24:
Yeah, I understand i made multiple mistakes. 

Regarding my order timeframe. I believe he 1.33 version was already in the latest status (it had a green tag). Which is why I just mindlessly ordered PCBs of this version.

I have a few more questions, but think those belong in the discussions tab or on discord.

I will now close this issue since, due to your explanation, i am unsure that the files in mention should really need editing.

Thank you for providing such a fast and thorough explanation of the release cycle!

closed 2023-12-24 by Catacomba

TweetyDaBird, 2023-12-24:
No problem!

FYI the 1.33 moved to 'latest release' at the same date as v1.40 was made as pre-release.