TweetyDaBird/Lotus-58-Classic #77: JLCPCB error with v1.28

U-Zd, 2024-03-20:
Hi,
Firstly, thanks for all your work!
I placed an order with the v1.28 gerber files and received the following email from JLCPCB engineer:

"Sorry to bother you, but there is an issue that we want to confirm with you before proceeding.
As shown below, the NPTH were connected with PTH, they will become PTH, we suggest to move those NPTH to ko layer to mill out?"

[image]

I am sorry, I do not know much about PCB designing so am unable to fix the issue myself
Thanks

TweetyDaBird, 2024-03-20:
Hi there. 

Thanks for the praise. 

As for the question from JLCPCB it's just them being dumb in the extreme, and suggesting the working way to 'fix' something that isn't an issue. 

Answer them this: "No, the PTH holes should not be moved/changed in any way. It's perfectly ok for the NPTH holes to become partially plated, and this should be manufactured as is specified and tested."

And for future reference if anyone is looking for the answer here this applies to all of their silly questions. The design is tested and already manufactured **as specified** by the time it makes it to a release.

TweetyDaBird, 2024-03-20:
Btw. I took the liberty of changing the headline to make it more universal for people searching for answers.

closed 2024-03-20 by TweetyDaBird