TweetyDaBird/Lotus-58-Classic #81: v1.29 TRRS issues: one key stuck & no R1/R2

ClemensSchartmueller, 2024-10-14:
Hi,

I have recently acquired two v1.29 TRRS boards from your shop with two OLEDs and through-the-hole soldering. So far, everything seems to work nicely, except:

1. one side has a key stuck (LH, bottom 3-key row, first key left-to-right) : It seems to me that I somehow shorted the hotswap socket (checked with multimeter and no switch inserted). Is there any fix to that beyond buying a new socket and resoldering it (sadly the order only had the exact amount of parts in it, not a single spare one for beginners like :D) ?
2. The manual says "solder R1, R2 for OLED"  and C1/C2 for an encoder. are the capacitors really needed for encoders? I tested without and it seems to work fine. Also: R1, R2 are not listed on the board - are these the same slots as C1/C2? 

Thanks so much in advance for your help!

TweetyDaBird, 2024-10-14:
Hi. Sorry but the kit even though for beginners is based on the assumption that it's really, really hard to mess up the components as such. 

I'm willing to be that the socket in itself is perfectly fine, and if you look carefully you have some left over tin somewhere causing the short on the pads/traces. To actually damage/short the socket as such, you'd have to drench it in solder and get it in places quite far from the solder pads. Not impossible but quite unlikely. (If you truly managed to short the socket and can't get the tin out, drop a note in the store, and I'll happily send you a socket, no charge).

Yeah. But the manual also say something at the very beginning that you obviously missed. It states that for every new version refer to the silkscreen as that is always more correct/updated than the manual. And since quite a few revisions back, there are no resistors as they are no longer needed. 

Capacitors aren't 'needed' for normal use and to make the encoders 'work'. They are there to prevent noise from the encoder to trigger false key inputs, or make inputs be missed, or even as a worst case reset the controller and restart the keyboard. You can certainly leave them out if you want too, but I'd say it's a small effort to put them in place not to have issues at random.

ClemensSchartmueller, 2024-10-14:
Thanks for the reply. I'll check again! I also removed any potential excess solder with a solder sucker but the problem persists. Do you have a hint what other points could be shorted to show such behavior? I checked the arduino pro micro too.

@R1/R2 and capacitors: thanks, that clarifies things!

TweetyDaBird, 2024-11-08:
Not sure where this ended up. (I have a hard time keeping names across platforms straight.)?

TweetyDaBird, 2025-04-01:
Closed ast there AFAIK are no design issues, and build/user questions belong in discord.

closed 2025-04-01 by TweetyDaBird