TweetyDaBird/Lotus-58-Classic #83: PDF with schematics for each release

chatelao, 2025-02-04:
I ♥️♥️♥️ my keyboard, especially for tinkering. It was my wheelchair to the first steps with KiCAD, learning to delete, move and autoroute a slim Corne42 derivative from it.

If you would add only the schematics to each version, it would be a great service for debugging and patching without KiCAD in the long term. - Just as a maintaince manual.

Not yet there, but working on a automated generation of the documentation on each tag for my projects. But the learning curve is SO steap.

TweetyDaBird, 2025-02-04:
I'll happily put it on the to-do list, but it's going to be low priority, so don't expect it anytime soon.

chatelao, 2025-02-04:
That's totally fine. It was just something I found out today and maybe adding it to new boards will be easy.

** Don't ask how I version my files 😂, sometimes JLC is my last hope.

closed 2025-04-02 by TweetyDaBird