TweetyDaBird/Lotus-58-Classic #85: Micro controller on the right half gets very hot quickly

tim-kilian, 2025-02-16:
Hej,

I've finished assembling the board and everything works ok, but the micro controller on the right side gets very hot. I tried already changing it, but both had the same issue. I have build the left half with Nice Nanos, a rotary encoder, a display and a battery and set it up with basic ZMK firmware with the left being the main board. Do you know what the issue could be?

TweetyDaBird, 2025-02-16:
Usually if things get hot and do not work, it's a short circuit. But since you are saying it does work that's not the case here. 

If I'd hazard a guess, is the OLED on the right side? And could it be the 3.3v regulator that is getting hot? The OLED screen does draw a bit more power than the n!n is really designed for, especially when powered of 3.3v as it's originally intended for 5v, so it would in-fact get at least lukewarm from that throughput. 

Other than that, I really have no good guesses.

TweetyDaBird, 2025-04-01:
Closed as there is no design error AFAIK.

closed 2025-04-01 by TweetyDaBird

tim-kilian, 2025-11-25:
I soldered the ESD the wrong way around. Now everything works fine. Thanks for the great keyboard !